margelo/react-native-vision-camera #3562: 🐛 iOS Build Errors with React Native Vision Camera on React Native 0.78.1 (Swift Concurrency Issues)

can-sevin, 2025-07-01:
### What's happening?

Hello everyone,

I am working on a project using React Native 0.78.1 and trying to integrate React Native Vision Camera. However, I am facing build errors on iOS that seem to be related to Swift concurrency and actor isolation in the CameraView.swift file. I need help figuring out how to resolve these issues.

Environment Information:

React Native: 0.78.1
React Native Vision Camera: 4.7.0 
Xcode: 16.2
iOS: 18.1

Has anyone encountered similar build errors with React Native 0.78.1 and React Native Vision Camera on iOS? Is there a known solution or workaround for these Swift concurrency-related issues? Any help or guidance would be greatly appreciated.

Thanks

### Reproduceable Code

```tsx
"react": "19.0.0",
    "react-native": "0.78.1",
    "react-native-vision-camera": "4.7.0",
```

### Relevant log output

```shell
❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:307:8: main actor-isolated instance method 'onError' cannot be used to satisfy nonisolated protocol requirement

  func onError(_ error: CameraError) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:327:8: main actor-isolated instance method 'onSessionInitialized()' cannot be used to satisfy nonisolated protocol requirement

  func onSessionInitialized() {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:331:8: main actor-isolated instance method 'onCameraStarted()' cannot be used to satisfy nonisolated protocol requirement

  func onCameraStarted() {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:335:8: main actor-isolated instance method 'onCameraStopped()' cannot be used to satisfy nonisolated protocol requirement

  func onCameraStopped() {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:347:8: main actor-isolated instance method 'onCaptureShutter(shutterType:)' cannot be used to satisfy nonisolated protocol requirement

  func onCaptureShutter(shutterType: ShutterType) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:353:8: main actor-isolated instance method 'onOutputOrientationChanged' cannot be used to satisfy nonisolated protocol requirement

  func onOutputOrientationChanged(_ outputOrientation: Orientation) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:359:8: main actor-isolated instance method 'onPreviewOrientationChanged' cannot be used to satisfy nonisolated protocol requirement

  func onPreviewOrientationChanged(_ previewOrientation: Orientation) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:365:8: main actor-isolated instance method 'onFrame(sampleBuffer:orientation:isMirrored:)' cannot be used to satisfy nonisolated protocol requirement

  func onFrame(sampleBuffer: CMSampleBuffer, orientation: Orientation, isMirrored: Bool) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:383:8: main actor-isolated instance method 'onCodeScanned(codes:scannerFrame:)' cannot be used to satisfy nonisolated protocol requirement

  func onCodeScanned(codes: [CameraSession.Code], scannerFrame: CameraSession.CodeScannerFrame) {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:339:8: main actor-isolated instance method 'onPreviewStarted()' cannot be used to satisfy nonisolated protocol requirement

  func onPreviewStarted() {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:343:8: main actor-isolated instance method 'onPreviewStopped()' cannot be used to satisfy nonisolated protocol requirement

  func onPreviewStopped() {
       ^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/React/CameraView.swift:390:8: main actor-isolated instance method 'onAverageFpsChanged(averageFps:)' cannot be used to satisfy nonisolated protocol requirement

  func onAverageFpsChanged(averageFps: Double) {
       ^

⚠️  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/Core/Types/Flash.swift:14:1: @frozen has no effect on non-public enums

@frozen
^

❌  /Users/can.sevin/novv/app-gntn/node_modules/react-native-vision-camera/ios/Core/Utils/GlobalReferenceHolder.swift:11:22: static property 'references' is not concurrency-safe because it is nonisolated global shared mutable state

  private static var references: [GlobalReferenceHolder] = []
```

### Camera Device

```json
I cannot reach the info.
```

### Device

Simulator iPhone 16 pro iOS 18.1

### VisionCamera Version

4.7.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kmlrmdhn, 2025-07-22:
[image]

i have same issue, in here

"dependencies": {
    "@homielab/react-native-auto-scroll": "^0.0.10",
    "@react-native-async-storage/async-storage": "^1.24.0",
    "@react-native-clipboard/clipboard": "^1.16.3",
    "@react-native-community/datetimepicker": "^8.4.2",
    "@react-native-community/netinfo": "^11.4.1",
    "@react-native-firebase/app": "^17.3.0",
    "@react-native-firebase/messaging": "^17.3.0",
    "@react-native-picker/picker": "^2.11.1",
    "@react-native/new-app-screen": "0.80.1",
    "@react-navigation/native": "^6.1.18",
    "@react-navigation/native-stack": "^6.9.10",
    "@sayem314/react-native-keep-awake": "^1.3.1",
    "date-fns": "^4.1.0",
    "deprecated-react-native-prop-types": "^5.0.0",
    "intl": "^1.2.5",
    "moment": "^2.30.1",
    "patch-package": "^8.0.0",
    "react": "19.1.0",
    "react-native": "0.80.1",
    "react-native-calendar-picker": "^8.0.5",
    "react-native-check-version": "^1.4.0",
    "react-native-countdown-component": "^2.7.1",
    "react-native-datepicker": "^1.7.2",
    "react-native-device-info": "^14.0.4",
    "react-native-file-viewer": "^2.1.5",
    "react-native-flatlist-slider": "^1.0.6",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.27.1",
    "react-native-image-picker": "^8.2.1",
    "react-native-image-zoom-viewer": "^3.0.1",
    "react-native-linear-gradient": "^2.8.3",
    "react-native-logs": "^5.3.0",
    "react-native-modal": "^14.0.0-rc.1",
    "react-native-modal-datetime-picker": "^18.0.0",
    "react-native-modal-selector": "^2.1.2",
    "react-native-network-logger": "^2.0.1",
    "react-native-paper": "^5.14.5",
    "react-native-qrcode-svg": "^6.3.15",
    "react-native-responsive-screen": "^1.4.2",
    "react-native-safe-area-context": "^5.5.1",
    "react-native-screens": "^4.11.1",
    "react-native-snap-carousel": "^3.9.1",
    "react-native-svg": "^15.12.0",
    "react-native-vector-icons": "^10.2.0",
    "react-native-version-check": "^3.5.0",
    "react-native-vision-camera": "^4.7.0",
    "react-native-webview": "^13.15.0",
    "throttle-debounce": "^5.0.2",
    "validator": "^13.15.15"
  },

my package json

did anyone fix it?

akyolburak, 2025-07-31:
Hi,
We're facing the same issue on our apps, is there any update on this?