margelo/react-native-vision-camera #3582: 🐛 Recording video with audio often triggers `unknown/unknown` error

MathiasWP, 2025-07-17:
### What's happening?

Before opening the camera i render a page where the user has to give access to both camera and microphone, so i'm not able to understand why this happens. The full error that gets thrown is the following:

```
Warning: [unknown/unknown]: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (561145187), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x15ecc17d0 {Error Domain=NSOSStatusErrorDomain Code=561145187 "(null)" UserInfo={AVErrorFourCharCode='!rec'}}} (caused by {"message":"Error Domain=AVFoundationErrorDomain Code=-11800 \"The operation could not be completed\" UserInfo={NSLocalizedFailureReason=An unknown error occurred (561145187), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x15ecc17d0 {Error Domain=NSOSStatusErrorDomain Code=561145187 \"(null)\" UserInfo={AVErrorFourCharCode='!rec'}}}","domain":"AVFoundationErrorDomain","details":{"NSUnderlyingError":null,"NSLocalizedDescription":"The operation could not be completed","NSLocalizedFailureReason":"An unknown error occurred (561145187)"},"code":-11800})
    at CameraView (<anonymous>)
```

My code is based on the example app from this library.

### Reproduceable Code

```tsx
/**
* Capture.tsx
*/
import * as React from 'react'
import { useRef, useState, useCallback, useMemo } from 'react'
import { StyleSheet, Text, View } from 'react-native'
import { Gesture, GestureDetector } from 'react-native-gesture-handler'
import type { CameraProps, CameraRuntimeError, PhotoFile, VideoFile } from 'react-native-vision-camera'
import {
  useCameraDevice,
  useCameraFormat,
  useMicrophonePermission,
} from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'
import { CONTENT_SPACING, CONTROL_BUTTON_SIZE, MAX_ZOOM_FACTOR, SAFE_AREA_PADDING, SCREEN_HEIGHT, SCREEN_WIDTH } from './Constants'
import Reanimated, { Extrapolation, interpolate, useAnimatedProps, useSharedValue, runOnJS } from 'react-native-reanimated'
import { useEffect } from 'react'
import { useIsForeground } from './hooks/useIsForeground'
import { CaptureButton } from './CaptureButton'
import { PressableOpacity } from 'react-native-pressable-opacity'
import { useIsFocused } from '@react-navigation/core'
import { usePreferredCameraDevice } from './hooks/usePreferredCameraDevice'
import { BoltIcon as BoltIconOutline } from 'react-native-heroicons/outline';
import { ArrowPathRoundedSquareIcon } from 'react-native-heroicons/outline';
import { BoltIcon as BoltIconSolid } from 'react-native-heroicons/solid';
import { MoonIcon as MoonIconOutline } from 'react-native-heroicons/outline';
import { useEditScreen } from './EditScreenContext';

const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera)
Reanimated.addWhitelistedNativeProps({
  zoom: true,
})

const SCALE_FULL_ZOOM = 3

export default function Capture() {
  const camera = useRef<Camera>(null)
  const [isCameraInitialized, setIsCameraInitialized] = useState(false)
  const zoom = useSharedValue(1)
  const isPressingButton = useSharedValue(false)
  const [isTakingPhoto, setIsTakingPhoto] = useState(false)
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()
  const { isInEditScreen } = useEditScreen()
  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back')
  const [enableHdr, setEnableHdr] = useState(false)
  const [flash, setFlash] = useState<'off' | 'on'>('off')
  const [enableNightMode, setEnableNightMode] = useState(false)
  const [targetFps, setTargetFps] = useState(30)
  const microphone = useMicrophonePermission()

  // Camera needs to stay active if we're using flash because it takes a while to take the photo
  const isActive = isFocused && isForeground && !isInEditScreen && (!isTakingPhoto || flash === 'on')
  // camera device settings
  const [preferredDevice] = usePreferredCameraDevice()
  let device = useCameraDevice(cameraPosition)
[... 629 lines cut ...]
  "sensorOrientation": "portrait",
  "minZoom": 1,
  "position": "back"
}
```

### Device

iPhone 12 mini - IOS 18.6

### VisionCamera Version

4.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

MathiasWP, 2025-07-17:
Extra information:

I'm making a snapchat like app, and after the user has recorded i play the video using react-native-video. While this video is playing i still have the Camera app mounted in a stack below while setting active to false, since this is recommended for better performance. Is it possible that there are some conflicts with audio here? Setting muted=true on the react-native-video component still triggers the error, but if i completely remove the Video renderer, then i am not able to trigger this error.

longnguyenegs, 2025-07-18:
It looks like you're experiencing the same issue: https://github.com/mrousavy/react-native-vision-camera/issues/3560

I ran into this as well while using react-native-vision-camera together with react-native-video. I was able to resolve it by adding the following prop:
>  <Video disableAudioSessionManagement={true}

Hope this helps!

cloud-github, 2025-07-25:
> > > It looks like you're experiencing the same issue: [#3560](https://github.com/mrousavy/react-native-vision-camera/issues/3560)
> > > I ran into this as well while using react-native-vision-camera together with react-native-video. I was able to resolve it by adding the following prop:
> > > > <Video disableAudioSessionManagement={true}
> > > 
> > > 
> > > Hope this helps!
> > 
> > 
> > What are your "react-native-vision-camera" and "react-native-video" versions ?
> 
> [@cloud-github](https://github.com/cloud-github) : "react-native-video": "6.15.0", "react-native-vision-camera": "4.6.4",

Sorry, that combination is not working for me. 

`"dependencies": {
    "@react-navigation/native": "^7.0.15",
    "@react-navigation/native-stack": "^7.3.21",
    "@react-navigation/stack": "^7.1.2",
    "react": "18.3.1",
    "react-native": "0.76.0",
    "react-native-gesture-handler": "^2.24.0",
    "react-native-safe-area-context": "^5.3.0",
    "react-native-screens": "4.4.0",
    "react-native-video": "6.15.0",
    "react-native-vision-camera": "4.6.4",
    "react-native-webview": "^13.13.2",
    "rn-fetch-blob": "^0.12.0",
    "uuid": "^11.1.0"
  },`

**Camera.js**

`import React, { useState, useRef, useEffect } from 'react';
import {
  View,
  Text,
  TouchableOpacity,
  StyleSheet,
  Alert,
  Dimensions,
  ActivityIndicator,
  Linking,
} from 'react-native';
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useMicrophonePermission,
} from 'react-native-vision-camera';

const { width, height } = Dimensions.get('window');

const CameraComponent = ({ onVideoRecorded }) => {
  const [isRecording, setIsRecording] = useState(false);
  const [recordedVideoPath, setRecordedVideoPath] = useState(null);
  const [isCameraInitialized, setIsCameraInitialized] = useState(false);
  const cameraRef = useRef(null);
  
  // Use the latest API - useCameraDevice instead of useCameraDevices
  const backDevice = useCameraDevice('back');
  const frontDevice = useCameraDevice('front');
  
  // Select back camera first, fallback to front
  const device = backDevice || frontDevice;
  
  const { hasPermission: hasCameraPermission, requestPermission: requestCameraPermission } = useCameraPermission();
  const { hasPermission: hasMicrophonePermission, requestPermission: requestMicrophonePermission } = useMicrophonePermission();

  useEffect(() => {
    checkPermissions();
[... 429 lines cut ...]
    right: 0,
    flexDirection: 'row',
    justifyContent: 'space-around',
    paddingHorizontal: 20,
  },
  button: {
    backgroundColor: 'rgba(0, 0, 0, 0.7)',
    paddingHorizontal: 20,
    paddingVertical: 10,
    borderRadius: 5,
  },
  buttonText: {
    color: 'white',
    fontSize: 16,
    fontWeight: 'bold',
  },
});

export default VideoPlayer;`

Recording and playback work correctly the first time. However, when attempting to record a second time, the recording does not function as expected. and throws this Camera error,

`Camera error: [unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-10868), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x280cdb2a0 {Error Domain=NSOSStatusErrorDomain Code=-10868 "(null)"}}]`

I would greatly appreciate your assistance. Thank you very much!

mikebouwmans, 2025-08-14:
> Guten Tag, Hans here 🍻
>
> I see you have quite a detailed report, which is good! However, it looks like you did not check if this issue can be reproduced in the VisionCamera Example app. Please try [...]

@mrousavy I'm also having this issue. Just pinging since Hans suggested it and hasn't been done before.

mikebouwmans, 2025-08-27:
So after some debugging I also came across this issue. Then saw that react-native-vide- did a update regarding audio management. For me applying this patch in react-native-video worked so Vision Camera did not throw the above error anymore. 

https://github.com/TheWidlarzGroup/react-native-video/commit/d2c92a1f3f6579aa6607389de8e51e4b75012f3b