margelo/react-native-vision-camera #3586: Tried to access a JS module before the React instance was fully set up. Calls to ReactContext#getJSModule should only happen once initialize() has been called on your native module.

Master-Antonio, 2025-07-21:
> If you see the exception, come from visioncamera. `com.mrousavy.camera.react.CameraDevicesManager`
> 
> `AndroidRuntime: at com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent(CameraDevicesManager.kt:102) 07-21 11:53:22.790 19288 19373 E AndroidRuntime: at com.mrousavy.camera.react.CameraDevicesManager1.invokeSuspend(CameraDevicesManager.kt:74) 07-21 11:53:22.790 19288 19373 E AndroidRuntime: at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33) 07-21 11:53:22.790 19288 19373 E AndroidRuntime: at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101) 07-21 11:53:22.790 19288 19373` ecc ecc 

 _Originally posted by @Master-Antonio in [#3585](https://github.com/mrousavy/react-native-vision-camera/issues/3585#issuecomment-3096698007)_

ddkuz, 2025-07-22:
+ 1 

Ran into this while upgrading react and react-native
info Fetching system and libraries information...
System:
  OS: Windows 11 10.0.26100
  CPU: (12) x64 11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz
  Memory: 1.78 GB / 15.78 GB
Binaries:
  Node:
    version: 22.17.1
    path: ~\AppData\Local\Temp\yarn--1753220538921-0.6584790179539757\node.CMD
  Yarn:
    version: 1.22.22
    path: ~\AppData\Local\Temp\yarn--1753220538921-0.6584790179539757\yarn.CMD
  npm:
    version: 11.4.2
    path: C:\Program Files\nodejs\npm.CMD
  Watchman: Not Found
SDKs:
  Android SDK:
    API Levels:
      - "34"
      - "35"
    Build Tools:
      - 33.0.1
      - 34.0.0
      - 35.0.0
    System Images:
      - android-35 | Google APIs Intel x86_64 Atom
      - android-35 | Google Play Intel x86_64 Atom
    Android NDK: Not Found
  Windows SDK:
    AllowAllTrustedApps: Enabled
    AllowDevelopmentWithoutDevLicense: Enabled
    Versions:
      - 10.0.16299.0
      - 10.0.17134.0
      - 10.0.17763.0
      - 10.0.19041.0
      - 10.0.20348.0
      - 10.0.22000.0
      - 10.0.22621.0
IDEs:
  Android Studio: AI-251.25410.109.2511.13752376
  Visual Studio:
    - 15.9.35027.97 (Visual Studio Community 2017)
    - 17.10.35027.167 (Visual Studio Community 2022)
Languages:
  Java: 17.0.11
  Ruby: Not Found
npmPackages:
  "@react-native-community/cli":
    installed: 19.1.0
    wanted: 19.1.0
  react:
    installed: 19.1.0
    wanted: 19.1.0
  react-native:
    installed: 0.80.1
    wanted: 0.80.1
  react-native-windows: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: Not found
  newArchEnabled: Not found

Done in 4.57s.

Master-Antonio, 2025-07-23:
I have news now.
Its because visioncamera with newArchEnabled:true, without the turbo modules crash, doesnt work.
The problem is that for now many libraries arent still compatible to the turbo modules, so that is a big problem, visioncamera should be compatible also to the old way...

mannoeu, 2025-07-24:
Same here using react-native-vision-camera 4.7.1 with frame processors and react-native 0.75.5

```
java.lang.IllegalStateException: Tried to access a JS module before the React instance was fully set up. Calls to ReactContext#getJSModule should only happen once initialize() has been called on your native module.
        at com.facebook.react.bridge.BridgeReactContext.getJSModule(BridgeReactContext:95)
        at com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent(CameraDevicesManager:102)
        at com.mrousavy.camera.react.CameraDevicesManager$1.invokeSuspend(CameraDevicesManager:74)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask:101)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
        at java.lang.Thread.run(Thread.java:1012)
```

Threads:
```
Thread 2 - main - (RUNNABLE)
        at android.os.BinderProxy.transactNative(BinderProxy.java:-2)
        at android.os.BinderProxy.transact(BinderProxy.java:662)
        at android.view.IWindowSession$Stub$Proxy.relayout(IWindowSession.java:1570)
        at android.view.ViewRootImpl.relayoutWindow(ViewRootImpl.java:9786)
        at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:3884)
        at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:3116)
        at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:10885)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1301)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1309)
        at android.view.Choreographer.doCallbacks(Choreographer.java:923)
        at android.view.Choreographer.doFrame(Choreographer.java:852)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1283)
        at android.os.Handler.handleCallback(Handler.java:942)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:226)
        at android.os.Looper.loop(Looper.java:313)
        at android.app.ActivityThread.main(ActivityThread.java:8762)
        at java.lang.reflect.Method.invoke(Method.java:-2)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)

Thread 7015 - Signal Catcher - (WAITING)
        at unknown method(unknown file)

Thread 7016 - HeapTaskDaemon - (WAITING)
        at unknown method(unknown file)

Thread 7017 - ReferenceQueueDaemon - (WAITING)
        at java.lang.Object.wait(Object.java:-2)
        at java.lang.Object.wait(Object.java:405)
        at java.lang.Object.wait(Object.java:543)
        at java.lang.Daemons$ReferenceQueueDaemon.runInternal(Daemons.java:251)
        at java.lang.Daemons$Daemon.run(Daemons.java:131)
        at java.lang.Thread.run(Thread.java:1012)

Thread 7018 - Jit thread pool worker thread 0 - (RUNNABLE)
        at unknown method(unknown file)

Thread 7019 - FinalizerDaemon - (WAITING)
        at java.lang.Object.wait(Object.java:-2)
        at java.lang.Object.wait(Object.java:405)
        at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:207)
        at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:228)
        at java.lang.Daemons$FinalizerDaemon.runInternal(Daemons.java:331)
        at java.lang.Daemons$Daemon.run(Daemons.java:131)
        at java.lang.Thread.run(Thread.java:1012)

Thread 7020 - FinalizerWatchdogDaemon - (WAITING)
        at java.lang.Object.wait(Object.java:-2)
        at java.lang.Object.wait(Object.java:405)
        at java.lang.Object.wait(Object.java:543)
        at java.lang.Daemons$FinalizerWatchdogDaemon.sleepUntilNeeded(Daemons.java:464)
        at java.lang.Daemons$FinalizerWatchdogDaemon.runInternal(Daemons.java:444)
[... 337 lines cut ...]
        at java.lang.Thread.run(Thread.java:1012)

Thread 7059 - CameraX-core_camera_1 - (WAITING)
        at jdk.internal.misc.Unsafe.park(Unsafe.java:-2)
        at java.util.concurrent.locks.LockSupport.park(LockSupport.java:341)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionNode.block(AbstractQueuedSynchronizer.java:506)
        at java.util.concurrent.ForkJoinPool.unmanagedBlock(ForkJoinPool.java:3466)
        at java.util.concurrent.ForkJoinPool.managedBlock(ForkJoinPool.java:3437)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:1623)
        at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:435)
        at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1071)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1131)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
        at java.lang.Thread.run(Thread.java:1012)

Thread 7060 - pool-6-thread-2 - (RUNNABLE)
        at com.facebook.react.bridge.BridgeReactContext.getJSModule(BridgeReactContext:95)
        at com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent(CameraDevicesManager:102)
        at com.mrousavy.camera.react.CameraDevicesManager$1.invokeSuspend(CameraDevicesManager:74)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask:101)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
        at java.lang.Thread.run(Thread.java:1012)
```

Thilanka-97, 2025-07-28:
- react-native: 0.78.0
- New Architecture: Disabled

**Temporary Fix: Crash issue with version `4.7.1` — Downgraded to `4.7.0`**

Initially installed version `4.7.1`, but encountered crashing issues. 
After downgrading to `4.7.0`, the app is stable and no longer crashes.

ddkuz, 2025-07-29:
I resolved this by moving the init {} in CameraDevicesManager.kt to initialize() function

  override fun initialize() {
    super.initialize()
    // Init cameraProvider + manager as early as possible
    coroutineScope.launch {
      try {
        Log.i(TAG, "Initializing ProcessCameraProvider...")
        cameraProvider = ProcessCameraProvider.getInstance(reactContext).await(executor)
        Log.i(TAG, "Initializing ExtensionsManager...")
        extensionsManager = ExtensionsManager.getInstanceAsync(reactContext, cameraProvider!!).await(executor)
        Log.i(TAG, "Successfully initialized!")
      } catch (error: Throwable) {
        Log.e(TAG, "Failed to initialize ProcessCameraProvider/ExtensionsManager! Error: ${error.message}", error)
      }
      sendAvailableDevicesChangedEvent()
    }
    cameraManager.registerAvailabilityCallback(callback, null)
  }

chanphiromsok, 2025-07-31:
I also had this issue

- Android , Old Arch,0.79.5

workbyken, 2025-08-05:
This is still broken, even on 4.7.0 for me.

I had to create a patch following [ddkuz](https://github.com/ddkuz) instructions (thank you)

stee1ix, 2025-08-26:
Facing this issue myself on Old Arch. Any solutions?

williamgurzoni, 2025-09-04:
It looks like the issue started [here](https://github.com/mrousavy/react-native-vision-camera/pull/3575/files).
And there is a PR with a potential fix [here](https://github.com/mrousavy/react-native-vision-camera/pull/3597).

josegiufrida, 2025-09-05:
@mohitGraffers @mannoeu 

For me, the solution was downgrade to v4.7.0.
I haven't seen any further problems. Development and production.

abdymm, 2025-10-06:
hey guys, any news regarding this issue? our crashlytics still reporting this, I've tried to reproduce this locally but I can't.

our Sentry also not reporting this crash

Versions:
```
"react-native": "0.78.3",
"react-native-vision-camera": "^4.6.4"
```

is upgrading `react-native-vision-camera` into the latest version will fix it? this issue is happening after we upgrade our `react-native` from 77 into 78

another question is this a silent crash?

jslok, 2025-10-06:
@abdymm try the patches from here. works for me. https://github.com/mrousavy/react-native-vision-camera/issues/3609

jslok, 2025-10-06:
@abdymm I think I did but honestly can't remember for sure.

MahmoudMabrok, 2025-12-28:
@mattijsf  i tried [v4.7.3](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.7.3) and still crash happening