margelo/react-native-vision-camera #3590: 🔧 Must be one of: UIManagerType.DEFAULT, UIManagerType.FABRIC [WrongConstant] val uiManager = UIManagerHelper.getUIManager(context, uiManagerType)

allemptysheep, 2025-07-23:
### How were you trying to build the app?

"react-native": "0.77.0",

I received an error message during build

### Full build logs

```tsx
2 errors, 0 warnings

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:lintDebug'.
> Lint found errors in the project; aborting build.
  
  Fix the issues identified by lint, or create a baseline to see only new errors.
  To create a baseline, run gradlew updateLintBaseline after adding the following to the module's build.gradle file:
  
android {
      lint {
          baseline = file("lint-baseline.xml")
      }
  }

  For more details, see https://developer.android.com/studio/write/lint#snapshot
  
  /Users/dapin/Desktop/HANSCO/frontend/HANSCO_TMS_MOBILE_FRONTEND/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:75: Error: Must be one of: UIManagerType.DEFAULT, UIManagerType.FABRIC [WrongConstant]
        val uiManager = UIManagerHelper.getUIManager(context, uiManagerType) ?: throw Error("UIManager not found!")
                                                              ~~~~~~~~~~~~~
  /Users/dapin/Desktop/HANSCO/frontend/HANSCO_TMS_MOBILE_FRONTEND/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:75: Error: Must be one of: UIManagerType.DEFAULT, UIManagerType.FABRIC [WrongConstant]
        val uiManager = UIManagerHelper.getUIManager(context, uiManagerType) ?: throw Error("UIManager not found!")
                                                              ~~~~~~~~~~~~~
  
     Explanation for issues of type "WrongConstant":
     Ensures that when parameter in a method only allows a specific set of
     constants, calls obey those rules.
  
  2 errors, 0 warnings

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.10.2/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD FAILED in 2m 31s
2670 actionable tasks: 2403 executed, 267 up-to-date
```

### Project dependencies

```json
"dependencies": {
    "@adrianso/react-native-device-brightness": "^1.2.7",
    "@babel/plugin-proposal-export-default-from": "^7.25.9",
    "@bravemobile/react-native-code-push": "^10.0.0",
    "@notifee/react-native": "^9.1.8",
    "@react-native-async-storage/async-storage": "^2.1.1",
    "@react-native-firebase/app": "22.4.0",
    "@react-native-firebase/messaging": "22.4.0",
    "@react-native-vector-icons/common": "^11.0.0",
    "@react-native-vector-icons/ionicons": "^7.4.0",
    "@react-navigation/native": "^7.0.14",
    "@react-navigation/native-stack": "^7.2.0",
    "@reduxjs/toolkit": "^2.5.1",
    "@tanstack/react-query": "^5.69.0",
    "@types/react-native-vector-icons": "^6.4.18",
    "axios": "^1.7.9",
    "babel-plugin-module-resolver": "^5.0.2",
    "react": "18.3.1",
    "react-native": "0.77.0",
    "react-native-background-actions": "^4.0.1",
    "react-native-blob-util": "^0.21.2",
    "react-native-bootsplash": "6.3.8",
    "react-native-bouncy-checkbox": "^4.1.2",
    "react-native-date-picker": "^5.0.10",
    "react-native-dotenv": "^3.4.11",
    "react-native-dropdown-picker": "^5.4.6",
    "react-native-fs": "^2.20.0",
    "react-native-geolocation-service": "^5.3.1",
    "react-native-gesture-handler": "^2.23.0",
    "react-native-get-location": "^5.0.0",
    "react-native-image-crop-picker": "^0.50.1",
    "react-native-image-editor": "^0.0.1",
    "react-native-keychain": "^10.0.0",
    "react-native-maps": "1.3.2",
    "react-native-permissions": "^5.2.5",
    "react-native-qrcode-svg": "^6.3.15",
    "react-native-reanimated": "^3.19.0",
    "react-native-restart": "^0.0.27",
    "react-native-safe-area-context": "^5.2.0",
    "react-native-screens": "^4.6.0",
    "react-native-sound": "^0.12.0",
    "react-native-static-safe-area-insets": "^2.2.0",
    "react-native-svg": "^15.12.0",
    "react-native-track-player": "^4.1.1",
    "react-native-uuid": "^2.0.3",
    "react-native-vision-camera": "^4.7.1",
    "react-native-webview": "^13.13.5",
    "react-redux": "^9.2.0",
    "redux-persist": "^6.0.0",
    "ts-node": "^10.9.2"
  },
```

### VisionCamera Version

4.7.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

egzon-mustafa, 2025-09-29:
same here? 
RN 0.78

sureshmgituser, 2025-10-08:
Same issue for me
I am using 

    "react-native-vision-camera": "^4.7.2",
    "react-native": "^0.77.0",