margelo/react-native-vision-camera #3601: 🔧

Velkuss89, 2025-08-14:
### How were you trying to build the app?

I had to upgrade my react-native app for Android API 36, so I update my project dependancy to

```
    "react": "19.1.0",
    "react-native": "0.81.0",
    "react-native-vision-camera": "^4.7.1",
```

Also

```
buildscript {
    ext {
        buildToolsVersion = "36.0.0"
        minSdkVersion = 24
        compileSdkVersion = 36
        targetSdkVersion = 36
        ndkVersion = "27.1.12297006"
        kotlinVersion = "2.1.20"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}
```

I installed all latest version packages and when I run 

`yarn run android //"android": "react-native run-android"`

I get error in console

e: file:///~/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewManager.kt:31:5 Return type mismatch: expected 'MutableMap<String, Any>?', actual 'Map<String, Any>'.

e: file:///~/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:192:20 Unresolved reference 'currentActivity'.

As a quick fix, I change 

CamerViewManager.kt in line 45 I added .toMutableMap()

and in CameraViewModule.kt val activity = reactApplicationContext.currentActivity as? PermissionAwareActivity

Then it works

But yeah, if that something you could fix in next version or let me know if I am doing something wrong

### Full build logs

```tsx
yarn run android
info Installing the app...
Downloading https://services.gradle.org/distributions/gradle-8.14.3-bin.zip
.............10%.............20%.............30%.............40%.............50%.............60%.............70%.............80%.............90%..............100%

Welcome to Gradle 8.14.3!

Here are the highlights of this release:
 - Java 24 support
 - GraalVM Native Image toolchain selection
 - Enhancements to test reporting
 - Build Authoring improvements

[... 276 lines cut ...]
```

### VisionCamera Version

4.7.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

as169m, 2025-08-17:
Im also getting similar error, is it because react native 0.81 is not compatible with react native vision camera

closed 2025-08-19 by Velkuss89

nottalha, 2025-08-27:
same error. 

`> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: file:///Users/talha/eshharSmartEstate/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewManager.kt:31:5 Return type mismatch: expected 'MutableMap<String, Any>?', actual 'Map<String, Any>'.
e: file:///Users/talha/eshharSmartEstate/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:192:20 Unresolved reference 'currentActivity'.
`