margelo/react-native-vision-camera #3602: > Task :react-native-vision-camera:compileDebugKotlin FAILED

as169m, 2025-08-17:
### What's happening?

Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.

### Reproduceable Code

```tsx
unable to build only
```

### Relevant log output

```shell
PS D:\Projects\> npx react-native run-android
* daemon not running; starting now at tcp:5037
* daemon started successfully
info Launching emulator...
error Failed to launch emulator. Reason: It took too long to start and connect with Android emulator: Pixel_7. You can try starting the emulator manually from the terminal with: D:\Android\Sdk/emulator/emulator @Pixel_7.
warn Please launch an emulator manually or connect a device. Otherwise app may fail to launch.
info Installing the app...
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at D:\Projects\\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-worklets-core:buildCMakeDebug[arm64-v8a]	\\
C/C++: Hard link from 'C:\Users\ashok\.gradle\caches\8.14.3\transforms\eb6d09c91acf843499bdca80e0a66fd6\transformed\react-android-0.81.0-debug\prefab\modules\reactnative\libs\android.arm64-v8a\libreactnative.so' to 'D:\Projects\\node_modules\react-native-worklets-core\android\build\intermediates\cxx\Debug\033436h6\obj\arm64-v8a\libreactnative.so' failed. Doing a slower copy instead.
C/C++: Hard link from 'C:\Users\ashok\.gradle\caches\8.14.3\transforms\eb6d09c91acf843499bdca80e0a66fd6\transformed\react-android-0.81.0-debug\prefab\modules\reactnative\libs\android.arm64-v8a\libreactnative.so' to 'D:\Projects\\node_modules\react-native-worklets-core\android\build\intermediates\cxx\Debug\033436h6\obj\arm64-v8a\libreactnative.so' failed. Doing a slower copy instead.

> Task :react-native-worklets-core:buildCMakeDebug[armeabi-v7a]
C/C++: ninja: Entering directory `D:\Projects\\node_modules\react-native-worklets-core\android\.cxx\Debug\033436h6\armeabi-v7a'
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/android/cpp-adapter.cpp:6:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiWorkletApi.h:14:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/android/../cpp/sharedvalues/WKTJsiSharedValue.h:8:
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:97:22: warning: exception of type 'const std::runtime_error &' will be caught by earlier handler [-Wexceptions]
C/C++:    97 |       } catch (const std::runtime_error &err) {
C/C++:       |                      ^
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:94:22: note: for type 'const std::exception &'
C/C++:    94 |       } catch (const std::exception &err) {
C/C++:       |                      ^
C/C++: 1 warning generated.
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletApi.cpp:1:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletApi.h:14:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/android/../cpp/sharedvalues/WKTJsiSharedValue.h:8:
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:97:22: warning: exception of type 'const std::runtime_error &' will be caught by earlier handler [-Wexceptions]
C/C++:    97 |       } catch (const std::runtime_error &err) {
C/C++:       |                      ^
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:94:22: note: for type 'const std::exception &'
C/C++:    94 |       } catch (const std::exception &err) {
C/C++:       |                      ^
C/C++: 1 warning generated.

> Task :react-native-vision-camera:compileDebugKotlin FAILED

> Task :react-native-worklets-core:buildCMakeDebug[armeabi-v7a]
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:3:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletApi.h:14:
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/android/../cpp/sharedvalues/WKTJsiSharedValue.h:8:
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:97:22: warning: exception of type 'const std::runtime_error &' will be caught by earlier handler [-Wexceptions]
C/C++:    97 |       } catch (const std::runtime_error &err) {
C/C++:       |                      ^
C/C++: D:/Projects/node_modules/react-native-worklets-core/android/../cpp/WKTJsiDispatcher.h:94:22: note: for type 'const std::exception &'
C/C++:    94 |       } catch (const std::exception &err) {
C/C++:       |                      ^
C/C++: In file included from D:/Projects/node_modules/react-native-worklets-core/cpp/WKTJsiWorkletContext.cpp:15:
[... 90 lines cut ...]
### Camera Device

```json
no output since its not even running
```

### Device

pixel virtual device 

### VisionCamera Version

4.7.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

yossularko, 2025-08-19:
#3605 issue still persist after updating code on: D:\ReactNativeApp\node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\react\CameraDevicesManager.kt

error:
```
Note: Recompile with -Xlint:deprecation for details.
e: file:///D:/AdyawinsaApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewManager.kt:31:5 Return type mismatch: expected 'MutableMap<String, Any>?', actual 'Map<String, Any>'.
e: file:///D:/AdyawinsaApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:192:20 Unresolved reference 'currentActivity'.
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction        
   > Compilation error. See log for more details
```

CaptainJeff, 2025-08-20:
@yossularko sorry what do you mean by 

```yes, it's fixed after following https://github.com/mrousavy/react-native-vision-camera/pull/3605 and https://github.com/user-attachments/files/21842645/react-native-vision-camera%2B4.7.1.patch```

You mean you downloaded the patch. And then modified `node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraDevicesManager.kt` to add https://github.com/mrousavy/react-native-vision-camera/pull/3605 (then ran patch package) to group these into a single patch?

Just wanted to make sure i understood what you meant.

Cause when I try that i get a ton of new compilation issues (related to the patch package)