margelo/react-native-vision-camera #3630: 🐛 Google Compliance Warning: 16KB Page Size Requirement (Android 15+) with react-native-vision-camera 4.6.4

sindhusingh, 2025-09-18:
### What's happening?

We’ve recently started receiving compliance warnings from Google Play regarding the 16KB page size requirement (Android 15+), which specifically points to Vision Camera. We’re currently using react-native-vision-camera version 4.6.4. I wanted to check if there’s a specific version available that addresses this issue, or if it’s still under investigation, and if there’s any recommended way to resolve it?

[image]

### Reproduceable Code

```tsx
Google Compliance Warning, So no code added
```

### Relevant log output

```shell
Google Compliance Warning, So no code added
```

### Camera Device

```json
Google Compliance Warning, So no code added
```

### Device

Android

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jslok, 2025-09-19:
4.7.0 added 16kb page support

jslok, 2025-09-21:
> > 4.7.0 added 16kb page support
> 
> Okay [@jslok](https://github.com/jslok) , so if we bump the version from 4.6.4 to 4.7.2, that should resolve it, right?

Try it and let us know :)

Jigisha820, 2025-10-17:
Thank you, It worked for me on Emulators. Has anyone tried on a Physical device which have a 16 KB page size?
ndkVersion = "27.1.12297006"
"react-native-vision-camera": "^4.7.2",
"react": "19.0.0",
"react-native": "0.79.2",

Acex187x, 2025-10-17:
@Jigisha820 It seems they don't exist yet 😅. That's why they're making us update the apps.

PWAronofsky, 2026-01-12:
I am also getting this error in our project where we need ndk 28. Is there a workaround for this?
```
Error: Failed to load VisionCamera C++ library!
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "__cxa_init_primary_exception" referenced by "/data/app/~ourpath/base.apk!/lib/arm64-v8a/libVisionCamera.so"...
	at java.lang.Runtime.loadLibrary0(Runtime.java:1090)
	at java.lang.Runtime.loadLibrary0(Runtime.java:1012)
	at java.lang.System.loadLibrary(System.java:1765)
	at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
	at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
	at com.facebook.react.ReactPackageHelper.getNativeModuleIterator(ReactPackageHelper.kt:32)
	at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.kt:39)
	at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1576)
	at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1547)
	at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1458)
	at com.facebook.react.ReactInstanceManager.lambda$runCreateReactContextOnNewThread$2(ReactInstanceManager.java:1226)
	at com.facebook.react.ReactInstanceManager.$r8$lambda$PrBhihCbbAFk4ZReAALGanVLCyc(Unknown Source:0)
	at com.facebook.react.ReactInstanceManager$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
	at java.lang.Thread.run(Thread.java:1119)
FATAL EXCEPTION: create_react_context
Process: com.our.app, PID: 32176
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "__cxa_init_primary_exception" referenced by "/data/app/~ourpath/base.apk!/lib/arm64-v8a/libVisionCamera.so"...
	at java.lang.Runtime.loadLibrary0(Runtime.java:1090)
	at java.lang.Runtime.loadLibrary0(Runtime.java:1012)
	at java.lang.System.loadLibrary(System.java:1765)
	at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
	at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
	at com.facebook.react.ReactPackageHelper.getNativeModuleIterator(ReactPackageHelper.kt:32)
	at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.kt:39)
	at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1576)
	at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1547)
	at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1458)
	at com.facebook.react.ReactInstanceManager.lambda$runCreateReactContextOnNewThread$2(ReactInstanceManager.java:1226)
	at com.facebook.react.ReactInstanceManager.$r8$lambda$PrBhihCbbAFk4ZReAALGanVLCyc(Unknown Source:0)
	at com.facebook.react.ReactInstanceManager$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
	at java.lang.Thread.run(Thread.java:1119)

```

```
    minSdkVersion = 28
    compileSdkVersion = 35
    targetSdkVersion = 35
    ndkVersion = "28.2.13676358"
```
```
    "react": "18.3.1",
    "react-native": "0.77.3",
    "react-native-vision-camera": "~4.7.3",
```

mrousavy, 2026-01-12:
Are you 100% sure that you are building with NDK 28?